SlashDevin/NeoGPS #96: U-blox examples not working

lights0123, 2018-07-31:
The examples [ublox.ino](https://github.com/SlashDevin/NeoGPS/blob/master/examples/ublox/ublox.ino) and [PUBX.ino](https://github.com/SlashDevin/NeoGPS/blob/master/examples/PUBX/PUBX.ino) both fail to compile. Specifically, in ublox.ino, lines 322 to 324 fail to compile on platforms that aren't 8-bit (I'm using an Arduino Zero, but these errors still appear when compiling for other boards).

    DEBUG_PORT << F("fix object size = ") << sizeof(gps.fix()) << '\n';
    DEBUG_PORT << F("ubloxGPS object size = ") << sizeof(ubloxGPS) << '\n';
    DEBUG_PORT << F("MyGPS object size = ") << sizeof(gps) << '\n';

This is because `sizeof()` returns a type that isn't compatible with Streamers.h. They should be casted to `uint32_t`s, or a new `operator` should be added to Streamers.h that accepts `size_t`.

However, after that issue is fixed, it fails with the error

```
/tmp/arduino_build_659507/sketch/ublox.ino.cpp.o: In function `ubloxGPS::enable_msg(ublox::msg_class_t, ublox::msg_id_t)':
/home/benschattinger/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:92: undefined reference to `ubloxGPS::send(ublox::msg_t const&, ublox::msg_t*)'
/tmp/arduino_build_659507/sketch/ublox.ino.cpp.o: In function `ubloxGPS::disable_msg(ublox::msg_class_t, ublox::msg_id_t)':
/home/benschattinger/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:97: undefined reference to `ubloxGPS::send(ublox::msg_t const&, ublox::msg_t*)'
/tmp/arduino_build_659507/sketch/ublox.ino.cpp.o: In function `configNMEA':
/tmp/arduino_modified_sketch_467266/ublox.ino:297: undefined reference to `ublox::configNMEA(ubloxGPS&, NMEAGPS::nmea_msg_t, unsigned char)'
/tmp/arduino_modified_sketch_467266/ublox.ino:297: undefined reference to `ublox::configNMEA(ubloxGPS&, NMEAGPS::nmea_msg_t, unsigned char)'
/tmp/arduino_build_659507/sketch/ublox.ino.cpp.o:(.rodata._ZTV5MyGPS[_ZTV5MyGPS]+0x8): undefined reference to `ubloxGPS::decode(char)'
/tmp/arduino_build_659507/sketch/ublox.ino.cpp.o:(.rodata._ZTV5MyGPS[_ZTV5MyGPS]+0x18): undefined reference to `ubloxGPS::parseField(char)'
```

This is because the [send() function](https://github.com/SlashDevin/NeoGPS/blob/1707c2c462e150fdb367bef6b6df51d14632b9a8/src/ublox/ubxGPS.h#L128) **hasn't been implemented**. Maybe you're missing a `using NMEAGPS::send;`? I don't understand the second error, because it seems as if that function is implemented properly.

---

The other problem is with PUBX.ino. After everything is compiled, during the linking stage, I get this error:

```
Linking everything together...
"/home/benschattinger/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc"  "-L/tmp/arduino_build_337502" -Os -Wl,--gc-sections -save-temps  "-T/home/benschattinger/.arduino15/packages/arduino/hardware/samd/1.6.19/variants/arduino_zero/linker_scripts/gcc/flash_with_bootloader.ld" "-Wl,-Map,/tmp/arduino_build_337502/PUBX.ino.map" --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m0plus -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o "/tmp/arduino_build_337502/PUBX.ino.elf" "/tmp/arduino_build_337502/sketch/PUBX.ino.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/DMS.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/GPSTime.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/Location.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/NMEAGPS.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/NeoTime.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/Streamers.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/Garmin/GrmNMEA.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/ublox/ubxGPS.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/ublox/ubxNMEA.cpp.o" "/tmp/arduino_build_337502/libraries/NeoGPS/ublox/ubxmsg.cpp.o" "/tmp/arduino_build_337502/core/variant.cpp.o" -Wl,--start-group "-L/home/benschattinger/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Lib/GCC/" -larm_cortexM0l_math -lm "/tmp/arduino_build_337502/../arduino_cache_913362/core/core_arduino_samd_arduino_zero_native_9bebe5f1758d050b1ce84adef3cfccdc.a" -Wl,--end-group
/tmp/arduino_build_337502/sketch/PUBX.ino.cpp.o: In function `_GLOBAL__sub_I_setup':
/tmp/arduino_modified_sketch_533537/PUBX.ino:158: undefined reference to `vtable for ubloxNMEA'
collect2: error: ld returned 1 exit status
Using library NeoGPS at version 4.2.9 in folder: /home/benschattinger/Arduino/libraries/NeoGPS 
```

Doom4535, 2018-09-09:
Have you been able to resolve this issue?  I'm having problems building ublxRate.ino, I did change ubxmsg.h line 180 from:

`enum time_ref_t time_ref:16;`

to:

`enum time_ref_t time_ref;`

And this gets me to:

`/tmp/cc8lyy9N.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x26d2): undefined reference to `ubloxGPS::write_P(ublox::msg_t const&)'
/tmp/cc8lyy9N.ltrans5.ltrans.o: In function `_GLOBAL__sub_I_LastSentenceInInterval':
<artificial>:(.text.startup+0x88): undefined reference to `vtable for ubloxGPS'
collect2: error: ld returned 1 exit status`

Which looks to be related to the same problem?

lights0123, 2018-09-10:
Nope, I haven't gotten anywhere with it - I've just been using standard NMEA messages.

Midnight28, 2018-10-28:
Hi all!

I also have the same problem, I can't even get PUBX.ino compiled, because I get this error message:

```
In function `global constructors keyed to 65535_0_PUBX.ino.cpp.o.2775':

<artificial>:(.text.startup+0xe2): undefined reference to `vtable for ubloxNMEA'

<artificial>:(.text.startup+0xe4): undefined reference to `vtable for ubloxNMEA'

collect2.exe: error: ld returned 1 exit status
```

I'm using an Arduino Mega with an ublox M8N module.
I did not spend time on investigating the problem, I just hope that the developers may see that many of us are interested in it :)

Another strange thing, that I've experienced with PUBX.ino: when I try to upload the example sketch I get this error:
```
PUBX:66:4: error: #error You must define NMEAGPS_EXPLICIT_MERGING in NMEAGPS_cfg.h

   #error You must define NMEAGPS_EXPLICIT_MERGING in NMEAGPS_cfg.h
```

Despite I have defined it before... But I can simply resolve this with commenting out the error message lines in PUBX.ino :)

Any help would be appreciated!

Midnight28, 2018-11-03:
OK, for me PUBX.ino and ublox.ino are working now!

I deleted the whole NeoGPS library and downloaded it again, started everything from the begining.

I work with VScode + Arduino extension. Here is what I did:

1. Choose the board (Arduino Mega), and the Baud rate (9600 by default)
2. Upload the NMEA.ino example first - it worked immediately without any troubles
3. Tried to upload the PUBX.ino example **WITHOUT any changes** in configuration files ( GPSfix_cfg.h, NeoGPS_cfg.h, NMEAGPS_cfg.h)
4. Of course it gave Error messages for the first time, asking for modifications in NMEAGPS_cfg.h - i.e. uncomment some lines, which I did
5. After uncommenting the necessary lines, I managed to compile and upload the code
6. I got an Error message on the Serial monitor first, what said 

> LAST_SENTENCE_IN_INTERVAL should be PUBX_00

7. So in NMEAGPS_cfg.h I changed the LAST_SENTENCE_IN_INTERVAL to this: 
`
#define LAST_SENTENCE_IN_INTERVAL (NMEAGPS::nmea_msg_t)(NMEAGPS::NMEA_LAST_MSG+1)
`
8. After that, the PUBX.ino started to work and print the fixes on the Serial monitor

I intentionally **didn't** use the NeoGPS/extras/configs/PUBX configuration, because the code didn't compiled with that, there were several errors.

pinnaclesystemsgroup, 2021-04-25:
Beginning a new project with this library and I want to know if this issue has been closed Thanks