SlashDevin/NeoGPS #105: fix.satellites is zero

trogper, 2018-09-09:
I'm using NEO m8n with arduino mkrfox1200. Location works fine, but fix.satellites is zero.
I have verified the NMEA sentences do contain satellite count using http://freenmea.net
Have tried Github version and also from arduino repository.
I have tried sending generated NMEA sentences to arduino via usb-uart, but it didn't work somehow.
I'm not posting the sentences due to security concerns.

Sekilsgs2, 2019-04-12:
Yes - always zero with neo6m.
Developer whats wrong? where satellites? You have tested example how get number of satellites?

trogper, 2019-04-12:
Actually it is not an issue of this library. I have not had GST messages enabled, which contain the count. If you use u.blox receiver, try configuring it using u-center.

closed 2019-04-12 by trogper