SlashDevin/NeoGPS #110: ESP32 HardwareSerial: Cannot receive any GPS data: Seems that gpsPort if flushed with gps.available()

AlexZibin, 2018-11-28:
All your library examples work fine on Arduino Nano; **thank you for your great job!**

But when I upload the sample code to ESP32 Dev Module, **gps.available( gpsPort ) always returns false** - simply saying, the code didn't work as expected. The same ESP32 hardware works fine with <Adafruit_GPS.h> examples - so I tried to find the reason in _NMEAGPS.cpp_, put some debugprints there, but didn't catch the bug.

Here is the sample code:

```
#define GPSport_h
#include <NMEAGPS.h>
NMEAGPS gps;
gps_fix fix;

#define DEBUG_PORT  Serial
#define gpsPort     Serial2
#define RXD2        16
#define TXD2        17

void setup() {
    Serial.begin(115200);
    while (!Serial) {
        ;
    }
    
    gpsPort.begin(9600, SERIAL_8N1, RXD2, TXD2);
}

void loop() { 

#define THIS_WORKS_FINE
#ifdef  THIS_WORKS_FINE    
    while (gpsPort.available()) {
        DEBUG_PORT.print(char(gpsPort.read())); // This branch prints out all GPS data as expected
    }
#else        
    if (gps.available( gpsPort )) {
        DEBUG_PORT.println ("This branch never happens :(");
        fix = gps.read();

    } else {
        DEBUG_PORT.println ("We always get here: gps.available( gpsPort ) == false");
        while (gpsPort.available()) {
            DEBUG_PORT.print(char(gpsPort.read()));
            DEBUG_PORT.println ("This branch ALSO never happens :("); // Seems that gpsPort if flushed with gps.available()
            DEBUG_PORT.print(' ');
        }
        DEBUG_PORT.println ();
        delay(500);
    }
    if (fix.valid.location ) {
    }
#endif    
}

```

Please help to find the solution :)

Vorms, 2018-12-22:
Hello,
I am interested to use your library with ESP32. Can you tell me if you find a workarround ?
Best regards
Thierry

Vorms, 2018-12-24:
I use the gpstiny library...
And you ?

AlexZibin, 2018-12-24:
I tried ada, it works.
But NeoGPS seems to be the most smart one - just need some bugfixes for ESP32.

mattogodoy, 2019-01-14:
Also interested in making it work with ESP32!

AlexZibin, 2019-01-14:
Also still interested in it.
Still hope the developer will answer us.

mattogodoy, 2019-01-14:
Invoking @SlashDevin 
Any ideas of why is this happening?

AlexZibin, 2019-01-15:
Need to dig deep into library source files.

cyberman54, 2019-02-09:
I cannot confirm this in general for ESP32. 
I'm using hardware Serial on a ESP32 board and it works with NeoGPS:

```
#define GPSport_h
#define DEBUG_PORT Serial

NMEAGPS gps;
gps_fix fix;

HardwareSerial GPS_Serial(1); // use UART #1
#define gpsPort GPS_Serial(1)
#define GPS_PORT_NAME "GPS_Serial 1"

GPS_Serial.begin(9600, SERIAL_8N1, GPIO_NUM_12, GPIO_NUM_15);

while (gps.available(GPS_Serial))
        fix = gps.read();
```

AlexZibin, 2019-02-11:
@cyberman54 
I tested your example on my board (just fixed typo in your line 8), but nothing changed.
I will try on different ESP32 board and report results later.

cyberman54, 2019-02-11:
@AlexZibin Where is the typo? i c&p'ed this from the running example, here it works.

AlexZibin, 2019-02-11:
@cyberman54 
`#define gpsPort GPS_Serial(1)`
didn't compile; so I changed it to 
`#define gpsPort GPS_Serial`

AlexZibin, 2019-02-11:
Just tried this code on bare ESP32-WROOM-32 chip with the same (bad) results.

timothy3001, 2019-03-19:
Any news on this?

AlexZibin, 2019-03-20:
No news at all. I keep using another lib for ESP32.

Rob4226, 2019-05-13:
I have the same issue on my ESP32 with HardwareSerial ports and the provided example programs!! FYI I am familiar with using this library successfully on Arduino's.  I found when I unplug the GPS modules VCC power line and then plug it right back in the gps.available() & fix.read() finally returns one or two results then goes back to not returning anything after that.  The same code works ok on an Arudino Nano but with the ESP32 I can't get the the code below to return any data either.  I tried using all three of the UART ports but none work like it should (aside from my trick of unpluging and repluging in the power of the GPS module which returns one or two GPS results).  I know the GPS is transmitting every second on the TX line because I can see LED attached to the RX pin on my ESP32 dev port and also the GPS outputs valid GPS sentence data when tested with a simple program on the same ESP32 board.  Any ideas on why this doesn't work??  Thanks.

```
while (gps.available())
        fix = gps.read();
```

Rob4226, 2019-05-16:
> I have the same issue on my ESP32 with HardwareSerial ports and the provided example programs!! FYI I am familiar with using this library successfully on Arduino's. I found when I unplug the GPS [...]

Ok, I figured it out.  Hopefully, this will help someone...Some GPS's like the Neo 6M series and also the one I am using right now that says "16E TTL from DIYmalls.com" on the label are u-blox based GPS's and use a different command to configure them.  Ublox NEO chips use "$PUBX" and the MTK chips use "$PMTK" commands.  It seems like most of the examples use "$PMTK" but I know there is a ublox folder.  Anyway the real problem why the gps.available() & fix.read() don't return anything is that by default, the NEO-6M outputs RMC, VTG, CGA, GSA, GSV, GLL messages once a second (these are standard NMEA messages) and for whatever reason it screws up those functions if you don't configure all the NeoGPS library options correctly.  The easiest way I found to fix this is to at setup send commands to shut off all the default NMEA sentences except the ones you need.  The below code shuts all off except the RMC sentence which is like the minimum amount of useful GPS information.  One I discovered to used "$PUBX" as the command prefix and to shut off some of the default output of the GPS module, the NeoGPS library's gps.available() & fix.read() sections of the code work fine.  Here is a small sketch that is a good starting point if you run into this situation and are using an ESP32 or any board using a HardwareSerial UART interface connecting the GPS module to the board...

```
/**
 * Rob 5/16/2019
 * Github: Rob4226
 * 
 * Basic code using U-blox based GPS module via UART 
 * using HardwareSerial and NeoGPS Arudino library to 
 * parse various NMEA sentences with ESP32 or other boards.
 * 
 * */

#include <Arduino.h>
#include <NMEAGPS.h>

//-------------------------------------------------------------------------
// Check that the config files are set up properly
#if !defined(NMEAGPS_PARSE_RMC) & \
    !defined(NMEAGPS_PARSE_GGA) & \
    !defined(NMEAGPS_PARSE_GLL)
#error You must uncomment at least one of NMEAGPS_PARSE_RMC, NMEAGPS_PARSE_GGA or NMEAGPS_PARSE_GLL in NMEAGPS_cfg.h!
#endif

#if !defined(GPS_FIX_LOCATION)
#error You must uncomment GPS_FIX_LOCATION in GPSfix_cfg.h!
#endif

#define gpsPort Serial2 //ESP32 pins 16RX, 17TX  UART02
#define GPS_PORT_NAME "ESP32_Serial2"
#define DEBUG_PORT Serial
#define BAUD_GPS 9600
#define BAUD_USB_SERIAL 115200
#define fixLEDPin 19

using namespace NeoGPS;
NMEAGPS gps;

void setup()
{
  Serial.begin(BAUD_USB_SERIAL);
  gpsPort.begin(BAUD_GPS);

  pinMode(fixLEDPin, OUTPUT);

  // Wait for USB Serial
  while (!Serial)
  {
    yield();
  }

  // By default, "16E TTL" and other U-blox GPS modules output RMC, VTG, CGA, GSA, GSV, GLL messages once a second (these are standard NMEA messages)
  // Configure the GPS to only output what is needed, like just RMC as is shown below by disabling all other default sentences.
  
  //gpsPort.println(F("$PUBX,40,RMC,0,0,0,0*47")); //RMC ON (Comented out to leave RMC on...uncomment this line to disable RMC)
  //delay(100);
  gpsPort.println(F("$PUBX,40,VTG,0,0,0,0*5E")); //VTG OFF
  delay(100);
  gpsPort.println(F("$PUBX,40,GGA,0,0,0,0*5A")); //CGA OFF
  delay(100);
  gpsPort.println(F("$PUBX,40,GSA,0,0,0,0*4E")); //GSA OFF
  delay(100);
  gpsPort.println(F("$PUBX,40,GSV,0,0,0,0*59")); //GSV OFF
  delay(100);
  gpsPort.println(F("$PUBX,40,GLL,0,0,0,0*5C")); //GLL OFF
  delay(100);
}

void loop()
{
  while (gps.available(gpsPort)) // basically true every second if 1Hz from GPS, only true when the GPS returns a sentence
  {
    gps_fix fix = gps.read(); // save the latest

    // Set the "fix" LED to on or off
    bool gpsWasFixed = fix.valid.status && (fix.status >= gps_fix::STATUS_STD);
    digitalWrite(fixLEDPin, gpsWasFixed);

    // When we have a valid location, print the latitude and longitude to the USB Serial terminal
    if (fix.valid.location)
    {
      DEBUG_PORT.println("The fix's location is valid!");
      DEBUG_PORT.print("     Latitude: ");
      DEBUG_PORT.println(fix.latitudeL());
      DEBUG_PORT.print("     Longitude: ");
      DEBUG_PORT.println(fix.longitudeL());
    }
  } // gps.available while loop
  
} // loop()
```

AlexZibin, 2019-05-16:
Wow that's great!
Thank you!

bamapookie, 2019-06-22:
Did you set the last message type?  I read somewhere in the code that it starts the quiet period and does the "hard work" once it knows it has received the last message.  If you have all the messages on, see which one comes last and set the last message type to that one.  See if that helps.

roblatour, 2020-08-15:
The above workaround is not working for me using an esp32 devkit v1 board.  

However, more or less the exact same code, with some changes for the rx and tx pin numbers, does work on an esp32-s2-saola-1 board.   

I've tried two esp32 devkit v1 boards, with the same non-functioning results.  Have quadruple checked the wiring.  Have tried powering the GPS unit and the esp32 devkit v1 board separately.

I've been battling this for a couple of days, any other suggestions for a workaround - or ideally a fix would be helpful.

roblatour, 2020-08-16:
So I eventually found a workaround as well; it follows the lead provided by Rob4226 above (thank you for that).

I was only after the GPS date and time,  so I edit the GPSfix_cfg.h file in the NeoGPS\src folder to only define date and time, as being needed and as follows:

#define GPS_FIX_DATE
#define GPS_FIX_TIME
//#define GPS_FIX_LOCATION
//#define GPS_FIX_LOCATION_DMS
//#define GPS_FIX_ALTITUDE
//#define GPS_FIX_SPEED
//#define GPS_FIX_VELNED
//#define GPS_FIX_HEADING
//#define GPS_FIX_SATELLITES
//#define GPS_FIX_HDOP
//#define GPS_FIX_VDOP
//#define GPS_FIX_PDOP
//#define GPS_FIX_LAT_ERR
//#define GPS_FIX_LON_ERR
//#define GPS_FIX_ALT_ERR
//#define GPS_FIX_SPD_ERR
//#define GPS_FIX_HDG_ERR
//#define GPS_FIX_TIME_ERR
//#define GPS_FIX_GEOID_HEIGHT

After that, it worked immediately.

of note, I also tested with
#define GPS_FIX_LOCATION
enabled (i.e. not commented out) and it also worked fine - I just don't need it for my app, so I will be leaving it commented out.

Hope this will be of help to others.

cbpdk, 2020-10-11:
I have tried to connect a GPS module to an esp32 (Nodecmu esp32) using HW Serial2.
This connection only worked from a cold boot, but hang up after a warm(sw) boot.
Using SoftwareSerial for a second serial poert, the connection always worked.

Searching for a solution, I came across one solution in the Arduino forum.

There is an explanation included:
[https://forum.arduino.cc/index.php?topic=605550.0](url)
see message #9

Here are snippets from my implementation adopted from the example code:
`
#include <NMEAGPS.h>
...

#define SERIAL2_RXGpio 16            //receive gpio UART2
#define SERIAL2_TXGpio 17            //transmit gpio UART2

#define gpsPort Serial2
#define GPS_PORT_NAME "Serial2"
#define DEBUG_PORT Serial
static const uint32_t GPSBaud = 9600;

...

void setup() {  

  //Serial2 not working correctly after .end() or warm boot
  gpio_matrix_out(SERIAL2_TXGpio, SIG_GPIO_OUT_IDX, false, false);
  pinMode(SERIAL2_TXGpio, OUTPUT);
  digitalWrite(SERIAL2_TXGpio, HIGH);
  delay(300);
  digitalWrite(SERIAL2_TXGpio, LOW);
  delay(25);
  digitalWrite(SERIAL2_TXGpio, HIGH);
  delay(25);

  
  Serial.begin(115200);
  gpsPort.begin(GPSBaud);  

...
`

tmargot, 2022-01-04:
Hopefully this may help someone, I appear to got mine working after hours of trying... 

**In GPSPort.h:**
I deleted everything and put:
```
#ifndef GPSport_h
#define GPSport_h

//           #define gpsPort Serial
             #define GPS_PORT_NAME "HardwareSerial"
             #define DEBUG_PORT Serial

#endif
```

**In .INO file:**
At the top included my includes and defined my required HardwareSerial variables. In setup changed the gpsPort,begin to include the defined HardwareSerial variables.
```
#include <HardwareSerial.h>
#include <GPSport.h>
#define GPS_SERIAL_NUM 1
#define GPS_RX_PIN 17
#define GPS_TX_PIN 16
HardwareSerial gpsPort(GPS_SERIAL_NUM);

void setup()
{
//normal code removed for clarity - changed gpsPort.begin( 9600) to below
gpsPort.begin( 9600, SERIAL_8N1, GPS_RX_PIN, GPS_TX_PIN);
}

```

R1DEN, 2023-07-05:
I'm gonna provide some more input to this one. I had a project since 2019 which is working perfectly fine on my old laptop with the old version of `toolchain-xtensa` and `framework-arduinoespressif32`. The platform version in PlatformIO is 1.11.1
On my new laptop with the 5.4.0 or 6.3.2 platform version - it doesn't work.
So the bug is NOT in the library itself, as it worked perfectly fine in the days it was created and maintained.
Unfortunately @SlashDevin's last commit in GitHub was in 2018 so we can only guess where he's now...

I would bet there's something changed in the logic how the HW serial port is being read in the newer framework which doesn't work well with @SlashDevin's original implementation.