SlashDevin/NeoGPS #127: Question on GPL3 Licence scope

qcabrol, 2019-07-04:
Hi,

I see that your work is licenced under GPL3. Though we would be interested to use it in a work including proprietary licenced elements as well.
Question is the scope of the licence: do we have the obligation to release all of our device firmware (strong copyleft - affecting the proprietary licenced parts of the project) or is it acceptable to release only the changes and improvements/extensions affecting the NeoGPS library?

Thanks in advance
Best

bamapookie, 2019-07-04:
I'm not an expert or a lawyer, but as I read the license, any modifications you make to the GPS software itself must also be released under the GPLv3.  If you use it inside of a larger product, that code doesn't need to be GPLv3.

qcabrol, 2019-07-05:
Ok, that was my concern. Would be nice to have this library under a more flexible Clear BSD, Apache2, MIT or even L-GPLv3 licensing. I guess I will rewrite my own as I need to support more of the UBX functions anyway.

bamapookie, 2019-07-06:
I doubt you are likely to get a license change.  @SlashDevin has not been active on GitHub (much less, this repo) for over a year.