SlashDevin/NeoGPS #142: NMEAdistance won't compile

mccgregory, 2020-08-01:
Trying to use NMEAdistance with an ESP32 but it fails to compile with the following error:
"NMEAdistance:54:51: error: call of overloaded 'Location_t(long int, long int)' is ambiguous"

It is referring to the line:
NeoGPS::Location_t base( -253448688L, 1310324914L ); // Ayers Rock, AU

If anybody could give me a pointer?

mccgregory, 2020-08-01:
The Latitude and Longitude in NMEAdistance are specified as LONG i.e.    ( -253448688L, 1310324914L ) 
But in **Location.h**,  **Location_t** is expecting two DOUBLEs.
here:
Location_t( double lat, double lon )
      : _lat(lat / LOC_SCALE), _lon(lon / LOC_SCALE)
        {}
Sorry, I'm not a programmer, how do I resolve this?
I can get rid of the compilation error by "getting the L out of it" (so to speak) with: ( -253448688, 1310324914 ).
But surely that will create a new problem?

bzroom, 2020-08-02:
try this:
//NeoGPS::Location_t base( -253448688L, 1310324914L ); // Ayers Rock, AU
NeoGPS::Location_t base;
base._lat = -253448688L;
base._lon =1310324914L; // Ayers Rock, AU

mccgregory, 2020-08-02:
HI, thanks for your time.

 

Fails with:

NMEAdistance:56:1: error: 'base' does not name a type

 

base._lat = -253448688L;

 

NMEAdistance:57:1: error: 'base' does not name a type

 

base._lon = 1310324914L; // Ayers Rock, AU

 

Best regards

Greg

bzroom, 2020-08-02:
You need to put those lines in a function, maybe setup() function. Cannot
put code outside of a function.

mccgregory, 2020-08-02:
Sorry, I’m a bit dim, and again thanks for your rapid feedback.

I don’t really understand…

You commented on the code, suggesting I try this:

showing the original line commented out, and three new lines.

I assumed I was just replacing that line with the three lines?

 

Are you suggesting I create another function to call the Location_t Function?

Apologies for my lack of understanding – I’m a bit old and thick!

Best regards

Greg

bzroom, 2020-08-02:
Post your whole code

mccgregory, 2020-08-02:
😊

My whole code is the NMEAdistance code.

At this stage I’m trying NeoGPS out, learning what can and can’t do.

It is a mini project in a bigger project – I just wanted something to display the accuracy of RTK GPS by displaying the distance between the BASE and the ROVER.

Regards

Greg

 

Post your whole code

bzroom, 2020-08-02:
Can you figure out how not to copy the whole thread every time you reply?

You need to put code inside a function. Ask a c++ programmer.

You being dim is not an NeoGPS issue. Try some c++ forums or Discord.

bzroom, 2020-08-02:
just use this one. FFS i gave you 2 examples:
NeoGPS::Location_t base( (int32_t)-253448688L, (int32_t)1310324914L ); // Ayers Rock, AU

mccgregory, 2020-08-02:
Thanks again.

just use this one. FFS i gave you 2 examples:
NeoGPS::Location_t base( (int32_t)-253448688L, (int32_t)1310324914L ); // Ayers Rock, AU

Sorry, the thread shows you gave me one example, which I commented on.

And then, now a new example – which appears to effective, it certainly compiles past that original error.

I now need to read some more of the NeoGPS documentation to fill out the other variables before I can fully compile it and then upload it and try it out.

Thanks for your time – hope your not too dislocated by your efforts.

Oh, and the answers to your previous questions:

Can you figure out how not to copy the whole thread every time you reply?

Sorry - Yes – how is this?

You need to put code inside a function. Ask a c++ programmer.

I sort of expected “Example Code” to work as is – more or less?

As far as I can see the NMEAdistance code, which after all, is meant to be example code, is calling an overloaded function Location_t with two long parameters. Location_t doesn’t seem to cope with longs.

You being dim is not an NeoGPS issue. Try some c++ forums or Discord.

My feeling the issue is probably that this code has been tested on a DUE or something like that not an ESP32, so that the variable type (float, double, long needs resolving)

I can probably resolve this myself – it just takes longer – I was merely reaching out…

The reason I looked on this particular forum was perhaps to find out if other people had experienced this and I could then gently nudge me in the right direction.

Apologies if I upset you or wasted your time.

Best regards

Greg

techbv, 2020-09-16:
Thank you very match, it help me.