SlashDevin/NeoGPS #144: error: reference to 'Print' is ambiguous

humanusdeus, 2020-08-30:
Can't compile the code examples, it always outputs the error:

`In file included from /home/warp/Arduino/libraries/NeoGPS/src/GPSfix.h:25:0,
                 from /home/warp/Arduino/libraries/NeoGPS/src/NMEAGPS.h:28,
                 from /home/warp/desktop/NeoGPS/examples/NMEAsimple/NMEAsimple.ino:1:
/home/warp/Arduino/libraries/NeoGPS/src/NeoTime.h:303:1: error: reference to 'Print' is ambiguous
 Print & operator <<( Print & outs, const NeoGPS::time_t &t );
 ^~~~~
/home/warp/Arduino/libraries/NeoGPS/src/NeoTime.h:295:7: note: candidates are: class Print
 class Print;
       ^~~~~
In file included from /home/warp/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Stream.h:25:0,
                 from /home/warp/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Client.h:22,
                 from /home/warp/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/ArduinoAPI.h:29,
                 from /home/warp/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/Arduino.h:23,
                 from /tmp/arduino-sketch-732DF1CC8158C71CF32499A187EEBBAB/sketch/NMEAsimple.ino.cpp:1:
/home/warp/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Print.h:34:7: note:                 class arduino::Print
 class Print
       ^~~~~`

daifukusensei, 2020-09-16:
Hi there,

Looking at the last line of your output, I'm experiencing the identical error.

I'm attempting to compile against a Nano Every.

My sketch runs fine on a Pro Micro and Mega.

Cheers.

daifukusensei, 2020-09-17:
I had to remove the 'class Print' forward declarations from the following files:
DMS.h
NeoTime.h

Also had to set SPI_DRIVER_SELECT to 1 per https://github.com/greiman/SdFat-beta/issues/42

Additionally, changed '#define F(str) (str)' in Arduino.h to '#define F(str) (const __FlashStringHelper*)(str)'

It at least now compiles, though I don't have my Every available until end of the week to test uploading and execution.

-Chris