SlashDevin/NeoGPS #147: NeoGPS::time_t issue

tvanderweide, 2021-01-14:
I'm trying to use the time_t struct to hold a dateTime from a fix and print this at a later point in the code. It's working fine except there's no dateTime_cs field in time_t and I had to do a little work around. Could this be added?

R1DEN, 2021-05-23:
AFAIK it is only available after the module receives the GPS "master frame" which is only "sent" once every 12.5 minutes. It sets the "leap second" property and that enables it to be accurate. I also tried to find a workaround on how to set it in a force way (my device has internet connection and can get the leap seond data from there)

R1DEN, 2021-05-23:
> I'm trying to use the time_t struct to hold a dateTime from a fix and print this at a later point in the code. It's working fine except there's no dateTime_cs field in time_t and I had to do a little work around. Could this be added or am I missing an easy way to access the sub-second GPS accuracy?
@tvanderweide 
AFAIK it is only available after the module receives the GPS "master frame" which is only "sent" once every 12.5 minutes. It sets the "leap second" property and that enables it to be accurate. I also tried to find a workaround on how to set it in a force way (my device has internet connection and can get the leap seond data from there)