SlashDevin/NeoGPS #148: ESP32 support

matburnham, 2021-03-14:
Is this library intended to work on anything but AVR?

I've gone down a bit of a rabbit hole and ended up finding out via SlashDevin/NeoHWSerial#13 that I'm going to struggle to get hardware Serial ports working on an ESP32.

matburnham, 2021-03-14:
Okay, it looks like that only applies if using interrupts. Which is what I was trying to do, but probably not vital.

So long as I don't define `NMEAGPS_INTERRUPT_PROCESSING` then `#define NEOGPS_USE_SERIAL1` works as expected.

sparkplug23, 2021-04-14:
Same goes for me. I would really love to get interrupts on an esp32 working with this. Did you manage it or come across another lib that manages it? This lib looks great but I really need to enable interrupts sadly.

ClimbingPenguin, 2021-04-15:
What I did as a subsitute / hack was to use the second core of my  ESP32 to do the gps parsing and waiting, and then the data would be passed on to the main core when it is available.

sparkplug23, 2021-04-15:
> What I did as a subsitute / hack was to use the second core of my  ESP32 to do the gps parsing and waiting, and then the data would be passed on to the main core when it is available.

@ClimbingPenguin Good idea. I actually have another receive serial process that needs priority over GPS, so I might use your method for that. 

Right now. I'm working on using freertos interrupts for UART together with ringbuffers, so similar to you, having it get the data ready in the background then passing the data to GPS (handle() function I assume?). I've got it working for GNRMC but the merging is being tricky for some reason. Early days yet though so I'll figure it out.

tmargot, 2022-01-04:
See my post, hopefully answers getting running on ESP32 to a certain extent 
https://github.com/SlashDevin/NeoGPS/issues/110#issuecomment-1004915803

R1DEN, 2023-07-05:
@sparkplug23 @tmargot you can check https://github.com/SlashDevin/NeoGPS/issues/110#issuecomment-1622238073