SlashDevin/NeoGPS #149: Cannot access 'fix.hdop' parameter!

saadi-tech, 2021-05-29:
When I try to access, fix.hdop using:
   hdop = fix.hdop;

(Screenshot attached)
[image]

I get the following error:
'class gps_fix' has no member named 'hdop'

(While I can easily access all other parameters like fix.latitude etc.

Please check into this issue and help me out. 

Thanks!

clavisound, 2021-05-31:
1. Project is abandoned.
2. The documentation is top-of-the-top: read it. https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Configurations.md
3. This is not a bug / issue.

danalvarez, 2021-08-14:
Do you have `#define GPS_FIX_HDOP` enabled in `GPSfix_cfg.h`? I think that must be enabled to be accessible via the fix structure.