SlashDevin/NeoGPS #150: And if GPS loose the fix?

RodEnry, 2021-06-16:
Hi SlashDevin, 
thank you for your library. A huge step forward respect the others you can find! 
The precision is great.

I'm facing the following issue.
Basically, I need to sample the GPS data at 10Hz on a SD card.
I'm switching on a led when the GPS has a fix through the followin function:

`void isGPSFix() {
    int GPSisFIX = 0; //Check if the GPS has a fix
    float currentLAT = 0;

    while (gps.available( GPS_Serial )) {
        currentFix = gps.read();
    } 

    GPSisFIX = currentFix.satellites;
    currentLAT = currentFix.location.latF();
    if ((GPSisFIX >= 3) && (currentLAT != 0)) {
        digitalWrite(ledPin, HIGH);
    } else {
        digitalWrite(ledPin, LOW);
    }

}`

This is working properly, unless I remove a cable form the GPS receiver.
In this case the data are basically lost and I can see a continus stream of identical values (last ones of the last fix).
Is there a way to check if some bad connection happens?

Many thanks
Enry

rthille, 2021-07-20:
Check the time of the latest fix and ensure it's newer than the previous fix, or that it's not too old relative to an external clock or 'millis()'?

danalvarez, 2021-08-14:
I think you need to check if the data is valid as shown in `NMEAsimple.ino`:

```
  while (gps.available( gpsPort )) {
    fix = gps.read();

    DEBUG_PORT.print( F("Location: ") );
    if (fix.valid.location) {
      DEBUG_PORT.print( fix.latitude(), 6 );
      DEBUG_PORT.print( ',' );
      DEBUG_PORT.print( fix.longitude(), 6 );
    }

    DEBUG_PORT.print( F(", Altitude: ") );
    if (fix.valid.altitude)
      DEBUG_PORT.print( fix.altitude() );

    DEBUG_PORT.println();
  }
```

This will only print lat, lon or altitude if the fix is valid (a.k.a. there is valid data coming in from your GPS).

svdrummer, 2021-10-29:
And the result was?

RodEnry, 2021-12-07:
Hello guys, 
I'm only checking if the satellites number was > 3 (3D fix) and if the LAT and LNG values were != 0.
@svdrummer

@danalvarez, honestly seems that the methods fix.valid.location are not working properly in my case. Any idea?
https://github.com/SlashDevin/NeoGPS/issues/155

danalvarez, 2021-12-07:
I would rather use `fix.status` to check how strong the fix is. `STATUS_DGPS` seems to be the highest possible, although I've gotten decent precision with `STATUS_STD`. More info here: https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Data%20Model.md

3 satellites isn't a particularly strong fix.